FYSETC/FYSETC-PortableInputShaper #1: Klipper.uf2

Randy1727, 2022-09-05:
I cant flash the device, Everytime I connect to PC and open folder as soon as I copy over Klipper.uf2 to it it disconnects and its not in there when I reconnect it.

d3d3d, 2022-10-07:
Same issue... connected to RASPi 4 not able to see the PIS mcu after flashing...
Something weird going on with the PIS bootloader and the USB interface state.

I found a workaround to be able to use device as intended (seems to work until the RASPi is rebooted).

Make klipper.uf2 in the default Klipper/out folder on your RASPi
While pressing the bootloader flash button on the PIS connect its USB cable to the RASPi.
From your favorite 'Terminal" and connected to the RASPi.
sudo mount /dev/sda1 /mnt
sudo cp /home/pi/klipper/out/klipper.uf2 /mnt/
sudo umount /mnt
After umount is completed,  ls -l /dev/serial/by-id/* should look something like this:
/dev/serial/by-id/usb-Klipper_rp2040_E66160F4231xxxxx-if00
Update your PIS serial: information in your cfg and continue to calibrate Input Shaper.
Seems to work until RASPi is rebooted, good thing this calibration is not needed too often.

I am able to run Input Shaper calibration macros successfully with the above...
Best of luck!
d3d3d

GerogeFu, 2022-10-09:
The pre-build klipper.uf2 firmware may not match your klipper version, to fix this, you need to build it on your own. Follow the instruction here https://www.klipper3d.org/Installation.html#building-and-flashing-the-micro-controller . And the firmware upload instruction steps are on Windows, do not drag the .uf2 file to pop-up folder, do copy action

d3d3d, 2022-10-09:
The FW Is being built properly, the same klipper.uf2 FW Version: v0.10.0-594-g7290c145 works properly on a Pico flashed on a Windows PC, Mac or RASPi. The Pico with only the mcu serial: info changed in the .cfg works every time using the same FW Versions, 3D printer, RASPi, power supplies, etc.

Below is copied from Klipper 'System Loads' while operational, showing the current firmware of all  three mcu's:
mcu(stm32f401xc) <-- E3S1Plus
Version: v0.10.0-594-g7290c145
mcu adxl(rp2040) <-- PIS
Version: v0.10.0-594-g7290c145
Host (armv7l) <-- RASPi 4
Version: v0.10.0-594-g7290c145

Perhaps I received a faulty PIS that ONLY works when FW is flashed while connected to the RASPi and then umount the PIS device. On the PIS I have, the USB interface is seen immediately after flashing and the device is unmounted. After a power cycle or a reboot the PIS USB interface is no longer seen. 

Since it seems you have not seen this issue before it is likely the PIS is defective. 
Will order another PIS expecting it to work.

Thanks!

geflores5, 2022-11-02:
I had the exact same issue of klipper not seeing the PIS. Ran the commands @d3d3d listed and it worked.

gogusrl, 2023-01-17:
I don't think it's related to the Pi, the gadget is not saving it's firmware over power offs. I have the same issue with a random pc running klipper and my windows desktop. 

If I boot with the button pressed and flash the firmware, it will reboot in serial mode and work properly. As soon as you remove the usb cable, it's gone and you need to flash it again. You should be able to reproduce this on any windows pc.

d3d3d, 2023-01-17:
gogusrl,

Since my last post Oct.2022,  I have replaced the faulty PIS with no further issues.

Although, I never implied the RasPi was causing the issue - you have figured out.
"the gadget (PIS) is not saving it's firmware over power offs" makes total sense.
PIS quality control?

The faulty PIS only worked for me because after "flashing" on the RasPi and umounting the device the PIS worked.
The faulty PIS worked after the umount command, but not after a power off or reboot.

Regards,

d3d3d

njsurf73, 2023-02-12:
same issue. has to be connected to the host, if flashed don any other computer it will not be recognized.

fryc88, 2023-04-23:
Same here with brand new FYSTEC PIP

fryc88, 2023-04-23:
With latest Klipper, apparently there is one more option available - Bootloader offset. When i changed it to 16KB it is working fine after unplugging and plugging back in into other devices, so that seemed to be an issue with older klipper versions, not rpi2040 chip itself.
[image]

njsurf73, 2023-04-23:
> With latest Klipper, apparently there is one more option available - Bootloader offset. When i changed it to 16KB it is working fine after unplugging and plugging back in into other devices, so that seemed to be an issue with older klipper versions, not rpi2040 chip itself. [image]

I did this and got a .bin not a .uf2 
Haven't loaded it yet, but that doesn't seem right

d3d3d, 2023-04-23:
njsurf73,

The RP2040 Bootloader offset should be set to 'No bootloader', the MAKE will then create the klipper.uf2 file.

RP2040 Bootloader operation is straight forward.

The RP2040 has a built-in first stage bootloader in ROM and can not be modified/updated.
The bootloader runs when the RP2040 is powered on while the bootloader button is pressed. 
The RP2040 is then in firmware flash mode.
When in firmware flash mode, the RP2040 flash is made visible as a mass storage device.
The operating system can then mount the RP2040 flash. 
After mounting, the kilpper.uf2 firmware can be copied to the RP2040 flash and should be retained.

One of the PIS devices I have is faulty, the RP2040 flash fails to retain the copied firmware after a power cycle OR reboot.

Hope this helps you or someone...

d3d3d

fryc88, 2023-04-23:
> I did this and got a .bin not a .uf2 Haven't loaded it yet, but that doesn't seem right

Ok, it appears to be my mistake. I'll double check it and reshare.