FYSETC/FYSETC-PortableInputShaper #2: The documentation could be improved a bit more

erlis, 2022-09-17:
Please just put 10 more mins on making the documentation a bit clear. The section for menu config is just an image... no steps? What am I supposed to do there? How I get there? Should I Open Menu config before or after flashing? 

It looks like this is very simple to setup but with the existing documentation is almost impossible to follow. 

Sorry I don't want to be negative but I'm all over the place finding how should I get this going and I feel this is the place where it should be done. 

Thanks

GerogeFu, 2022-11-09:
I just updated, have a check

erlis, 2022-11-09:
yes, that's better! Thanks for taking the time!

ahoy, 2022-11-13:
Could the documentation be updated? The menuconfig now has a bootloader option.

trashyman, 2023-02-24:
please specify if the bootloader needs to be set, mine can only be flashed via the RPi as mentioned in another issue-ticket.

Velo555, 2023-08-21:
Can we add a step file or pdf file with mechanical drawings? Wanted to design some mounts for a friend who does not live in the same state.

tastyratz, 2023-09-26:
Documentation is still very lacking, I had to stumble around to figure some things out:

1. a new shaper starts in flash mode, holding down the button does nothing till AFTER it's been flashed at least once
2. You cannot find out the Serial-ID when it's in usb flash/thumb drive mode
3. Dropping the files on the "thumb drive" flashes the device. You can get them with WINSCP after you build with make. Trying to do this from the PI sucks if you don't know linux
4.  Bootloader options should be left at off/unchecked, screenshot is outdated
5. "You can remove PIS after you get the [input_shaper] section parameter." This doesn't say enough on it's own. You MUST comment out that include if you unplug PIS or Klipper will not boot!
6.  The PIS.cfg file should have a callback and reference to the github in it. This is just commentary.
7. The commands are in other git issues but it would be helpful to just have them quick in the instructions over only linking the longer klipper doc. That's a great idea to include in case it changes.

I found a lot of things to do differently based on reported issues here. If I just had the instructions, I would have returned it/tossed the unit.