FYSETC/FYSETC-PortableInputShaper #4: Klipper wont Boot withouth PIS connectet

7R0J4Ner, 2023-01-23:
Hello

I got the PIS installed and everything works fine.
But when i remove the PIS and reboot Klipper, it wont Boot and an Error shows "Not able to connect to MCU" or something.

Does the PIS now always have to remain connected to the printer, or did I simply do something wrong during setup?

Thank you

dhcorley, 2023-02-25:
If you used the instructions. You just comment out the include in printer.cfg; and restart klipper. If your not permanently mounting it.

#include pis.cfg

DaVinci-10, 2023-02-25:
If you used the instructions. You just comment out the include in printer.cfg; and restart klipper. If your not permanently mounting it.

#include pis.cfg

DaVinci-10, 2023-03-01:
You can remove PIS after you get the [input_shaper] ...
https://github.com/FYSETC/FYSETC-PortableInputShaper/blob/5992485bd487227db958bc60abdc5699daa84f9c/README.md

SrgntBallistic, 2023-10-14:
In the official Klipper IS docs that they link to in the readme it says to comment out the include line when the adxl is disconnected

[image]

https://www.klipper3d.org/Measuring_Resonances.html