FYSETC/FYSETC-PortableInputShaper #5: Connection issue

njsurf73, 2023-02-09:
When I copy the firmware to the rp2040 it will not be recognized when I unplug and move to my PC running klipper. In order to make it work, I have to connect to the PC running klipper and then copy the firmware to the rp2040. And every time I restart I have to copy the firmware to the 2040 again on the PC I am using to run klipper.

DaVinci-10, 2023-02-25:
What is the os on the PC running klipper?
I am currently using Debian desktop version to run klipper. And when logged in. I first stop klipper. Unplug usb, hold the boot button while plugging in usb. It pops a box on the laptop asking if it should open the drive in Files app. Then I just navigate to klipper/out folder copy klipper.uf2 and paste onto the rp2040 drive...( copy paste, dont drag and drop)
 instantly the whole drive disappears...
Restart klipper and done. I've done it twice now no issues.

njsurf73, 2023-02-25:
My issue is, I have 3 printers running on one laptop. If I restart the laptop, I have to copy the firmware to the 2040 allover again to have it recognized. Otherwise it doesn't show back up as a mcu

DaVinci-10, 2023-02-26:
What shows up in a lsusb, or ls /dev/serial/by-id/ 
also can you also post your pis.cfg, maybe even a klippy.log
mainsail or fluidd? 
which printer is configured to use it? 7125 or higher or all and you just uncomment the one you have it on... ?

njsurf73, 2023-02-26:
I uncomment the printers not in use. I will post a log when I get home.

fryc88, 2023-04-23:
[image]

That setting, bootloader offset apparently solved my issue like yours, with disappearing device after connecting it from pc to my rpi with klipper. When this manual has been made, apparently klipper haven't had this option in make menuconfig.

njsurf73, 2023-04-23:
When I do this I get a .bin not a .uf2 file in klipper/outDid you change that or just load that

DaVinci-10, 2023-04-23:
the current klipper makemenu creates the .uf2 after compiling the .bin for the 2040
[image]

Version: v0.11.0-180-g011b4e39
  Building rp2040 stage2 out/stage2.o
  Preprocessing out/src/rp2040/rp2040_link.ld
  Linking out/klipper.elf
  Building out/lib/rp2040/elf2uf2/elf2uf2
  Creating uf2 file out/klipper.uf2
[✓ OK] Firmware built!

njsurf73, 2023-04-24:
> 

No matter how I build it, when I select the bootloader offset it only compiles a .bin
Using the makemenu and kiauh 
It will only build it to .bin

DaVinci-10, 2023-04-25:
bootloader offset should be set to none...
[image]
should look like this

```G3rKo — Today at 6:14 PM
CD klipper, make menuconfig,  select RP2040 ,no bootloader, connection usb,
Esc, yes, 
make -j4
should compile out/klipper.uf2
If select bootloader then no uf2 file will be generated.!
and you may need write access  "sudo cp out/klipper.uf2 /media..../"
And in PIS.config u need to replace the serial=/dev/serial/by-id/   with your id
get the id by "ls /dev/serial/by-id"```
from a discord discussion... 

can you capture this kiauh screen

```Version: v0.11.0-180-g011b4e39
  Building rp2040 stage2 out/stage2.o
  Preprocessing out/src/rp2040/rp2040_link.ld
  Linking out/klipper.elf
  Building out/lib/rp2040/elf2uf2/elf2uf2
  Creating uf2 file out/klipper.uf2
[✓ OK] Firmware built!
/=======================================================\
|     ~~~~~~~~~~~~~ [ Advanced Menu ] ~~~~~~~~~~~~~     |
|-------------------------------------------------------|
| Klipper & API:          | Mainsail:                   |
|  1) [Rollback]          |  6) [Theme installer]       |
|                         |                             |
| Firmware:               | System:                     |
|  2) [Build only]        |  7) [Change hostname]       |
|  3) [Flash only]        |                             |
|  4) [Build + Flash]     | Extras:                     |
|  5) [Get MCU ID]        |  8) [G-Code Shell Command]  |
|-------------------------------------------------------|
|                       B) « Back                       |
\=======================================================/
####### Perform action:```

njsurf73, 2023-04-25:
> bootloader offset should be set to none... [image] should look like this
> 
> ```
> CD klipper, make menuconfig,  select RP2040 ,no bootloader, connection usb,
> Esc, yes, 
> make -j4
> should compile out/klipper.uf2
> If select bootloader then no uf2 file will be generated.!
> and you may need write access  "sudo cp out/klipper.uf2 /media..../"
> And in PIS.config u need to replace the serial=/dev/serial/by-id/   with your id
> get the id by "ls /dev/serial/by-id"```
> from a discord discussion... 
> 
> can you capture this kiauh screen
> 
> ```Version: v0.11.0-180-g011b4e39
>   Building rp2040 stage2 out/stage2.o
>   Preprocessing out/src/rp2040/rp2040_link.ld
>   Linking out/klipper.elf
>   Building out/lib/rp2040/elf2uf2/elf2uf2
>   Creating uf2 file out/klipper.uf2
> [✓ OK] Firmware built!
> /=======================================================\
> |     ~~~~~~~~~~~~~ [ Advanced Menu ] ~~~~~~~~~~~~~     |
> |-------------------------------------------------------|
> | Klipper & API:          | Mainsail:                   |
> |  1) [Rollback]          |  6) [Theme installer]       |
> |                         |                             |
> | Firmware:               | System:                     |
> |  2) [Build only]        |  7) [Change hostname]       |
> |  3) [Flash only]        |                             |
> |  4) [Build + Flash]     | Extras:                     |
> |  5) [Get MCU ID]        |  8) [G-Code Shell Command]  |
> |-------------------------------------------------------|
> |                       B) « Back                       |
> \=======================================================/
> ####### Perform action:```
> ```

Yes. I can do that fine and get the correct firmware, but once I flash it to the pis, unless it is connected to my host it won't be recognized. If I flash it on the host, as soon as I restart it disappears. I was trying the boot offset because it was recommended as a fix.

DaVinci-10, 2023-04-25:
(sorry if this sounds over simplified, but the flashing procedure is simple file copy and paste, you dont flash it per se. ie you dont use the standard flash procedure like you do the printer mcu. 
It's easiest to do on a desktop flavor of linux, because as soon as you plug the usb in while holding the boot button down, A Jump (flash) drive pops up on the display, simply open and copy and paste the file from home/klipper/out. And as soon as you copy the file, the 2040 automaticly flashes itself internally and reboots. thats why the flash drive disappears.

njsurf73, 2023-05-13:
> (sorry if this sounds over simplified, but the flashing procedure is simple file copy and paste, you dont flash it per se. ie you dont use the standard flash procedure like you do the printer mcu. [...]

yes, i have used a pico and many varients of an rp2040. i understand how to flash it. The problem is once flashed it will be ok as long as its flashed from the host and remains powered on and plugged in. once it loses power or is unplugged, it reverts to not connecting as an mcu and has to be flashed again. there is a fault with the PIS not the process.

DaVinci-10, 2023-05-13:
Ok, I get that... so my next quesiton is:
After a hard boot, does it appear as a drive on the host like you held the boot button?
and more for the developer I guess, but does a soft reset after a flash ignore the boot button... ie would stuck boot button exibit that behavior

njsurf73, 2023-05-13:
No. It shows up on lsusb but as a connected device. It doesn't show up by /serial. Have to hold the boot button down again and reconnect to get it flashable.

SrgntBallistic, 2023-10-14:
I'm not having issues with the PIS not connecting
I just move the firmware over to the pico from the Pi

Disconnect the pico
Reconnect the pico while holding the boot button
run `> lsusb`
The Pico should show up as "RP2 Boot"
[image]
 cd into Klipper
` > make menuconfig`
` > make`
` > sudo mount /dev/sda1 /mnt`
` > sudo cp out/klipper.uf2 /mnt`
` > sudo umount /mnt`
` > ls /dev/serial/by-id`
[image]

That sticks for me and I can reconnect and rerun IS

This at least takes the process of connecting to a PC out of the flow

njsurf73, 2023-10-14:
Unfortunately it doesn't stick for me. I have to reflash every time it disconnects. I put together my own, and it's working as expected