FYSETC/FYSETC-PortableInputShaper #8: Unable to open serial port

Ltek, 2023-04-14:
Followed the instructions 100% but no matter what I try it doesn't connect (also tried multiple cables also; it flashed fine with the first cable). Klippy.log errors below.

PIS.cfg...

`[mcu PIS]
# Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
serial: /dev/serial/by-id/usb-Klipper_rp2040_E66118F5D79E8536-if00

[adxl345]
cs_pin: PIS:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: PIS:gpio10
spi_software_mosi_pin: PIS:gpio11
spi_software_miso_pin: PIS:gpio12
axes_map: x,-z,y

[resonance_tester]
accel_chip: adxl345
probe_points:
    155,155,20  # first two numbers are middle of bed`

Klippy.log...
`mcu 'mcu': Starting serial connect
Loaded MCU 'mcu' 105 commands (v0.11.0-173-ga8b1b0ef / gcc: (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027] binutils: (2.35.2-2+14+b2) 2.35.2)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c0a=gpio0,gpio1 BUS_PINS_i2c0b=gpio4,gpio5 BUS_PINS_i2c0c=gpio8,gpio9 BUS_PINS_i2c0d=gpio12,gpio13 BUS_PINS_i2c0e=gpio16,gpio17 BUS_PINS_i2c0f=gpio20,gpio21 BUS_PINS_i2c0g=gpio24,gpio25 BUS_PINS_i2c0h=gpio28,gpio29 BUS_PINS_i2c1a=gpio2,gpio3 BUS_PINS_i2c1b=gpio6,gpio7 BUS_PINS_i2c1c=gpio10,gpio11 BUS_PINS_i2c1d=gpio14,gpio15 BUS_PINS_i2c1e=gpio18,gpio19 BUS_PINS_i2c1f=gpio22,gpio23 BUS_PINS_i2c1g=gpio26,gpio27 BUS_PINS_spi0a=gpio0,gpio3,gpio2 BUS_PINS_spi0b=gpio4,gpio7,gpio6 BUS_PINS_spi0c=gpio16,gpio19,gpio18 BUS_PINS_spi0d=gpio20,gpio23,gpio22 BUS_PINS_spi1a=gpio8,gpio11,gpio10 BUS_PINS_spi1b=gpio12,gpio15,gpio14 BUS_PINS_spi1c=gpio24,gpio27,gpio26 CLOCK_FREQ=12000000 MCU=rp2040 PWM_MAX=255 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
mcu 'PIS': Starting serial connect
mcu 'PIS': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_rp2040_E66118F5D79E8536-if00: [Errno 11] Resource temporarily unavailable
webhooks client 548191637168: New connection
webhooks client 548191637168: Client info {'program': 'Moonraker', 'version': 'v0.8.0-35-geff0901'}
mcu 'PIS': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_rp2040_E66118F5D79E8536-if00: [Errno 11] Resource temporarily unavailable
mcu 'PIS': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_rp2040_E66118F5D79E8536-if00: [Errno 11] Resource temporarily unavailable
mcu 'PIS': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_rp2040_E66118F5D79E8536-if00: [Errno 11] Resource temporarily unavailable
mcu 'PIS': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_rp2040_E66118F5D79E8536-if00: [Errno 11] Resource temporarily unavailable
mcu 'PIS': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_rp2040_E66118F5D79E8536-if00: [Errno 11] Resource temporarily unavailable`

DaVinci-10, 2023-04-15:
can you confirm it does show up in a 
`ls -l /dev/serial/by-id/ `

Ltek, 2023-04-15:
It was odd but after multiple reboots (Pi & printer) it started connecting
fine.

gabriellferreira, 2023-05-14:
just as a follow-up from this thread
I was also having issues, then I found a comment in the [amazon product](https://www.amazon.com/dp/B0B8MMW2Z9/ref=cm_cr_dp_d_vote_lft?ie=UTF8&csrfT=hJhQyngdabosrjrDIT17tKW1hi05rm4BDse3C2g0etKIAAAAAGRhPWoAAAAB&reviewId=R2W18RGW1VIAX3#R2W18RGW1VIAX3) page that finally made it work here

```

The Fysetc instructions on Github are not complete and unless you are a pro you may have issues getting it set up.

With the help from klipper discord these instructions helped me get it set up.

MAKE SURE YOUR KLIPPER IS UP-TO-DATE

Via SSH to your PI with Klipper installed
1. Change directory to Klipper
2. make menuconfig
3. set menu per image on Fysetc PIS Github
3.1 Do not enable extra low-level configuration options
3.2 Set Micro-controller Architecutre to Raspberry PI RP2040
3.3 Set communication interface to USB
4. Exit and save menu config.
5. sudo service klipper stop
6. make flash
7. sudo service klipper start
Now the firmware 'klipper.uf2' should be created and located on your pi in the klipper/out directory

FTP into your PI
1. Navigate to the Klipper/Out directory
2. Download the 'klipper'uf2' file onto your pc desktop.
3. Exit and quit FTP

Now flash the firmware 'klipper.uf2' onto the Fysetc PIS
1. Press and hold the button on the Fysetc PIS
2. Connect the PIS with USB C to your PC
3. Release the button on the PIS
4. A folder named RPI-RP2 will pop up
5. Copy/Move the 'klipper.uf2' file from your desktop into the RPI-RP2 folder
6. Wait - the folder should close automatically
7. Unplug the PIS from your PC

Now set up PIS on your PI via USB
1. Press and Hold the button on the PIS
2. Connect PIS to your Raspberry PI

Now set up the firmware
1. Via SSH issue - lsusb
2. locate the device ID for the RP2040 -- i.e. Bus 001 Device 008: ID 2e8a:0003 Raspberry Pi RP2 Boot
3. Copy the ID i.e. '2e8a: 0003'
4. CD to your Klipper directory
5. issue command- make flash FLASH_DEVICE=2e8a:0003 << CHANGE TO YOUR DEVICE ID
6. You should see that the UF2 has been flashed

Now lets find the serial address for the Fysetc PIS RP2040 connected to your PI
1. issue command - ls /dev/serial/by-id/*
2. copy the address denoted with rp2040 -- ie. /dev/serial/by-id/usb-Klipper_rp2040_E66160F4235E4038-if00

Now lets create the PIS.cfg in klipper
1. Machine >> Create File >> name it PIS.cfg OR download it from Fysetc git up and upload the file
2. Copy contents from the Fysetc PIS file into your kipper PIS.cfg
3. Edit the "serial" line to have your path you copied from the prior steps for obtaining the serial address
4. Save/close/restart Klipper on your PI
5. Edit your printer.cfg to add a line that includes the PIS.cfg -- [include PIS.cfg]
6. Save/close/restart Klipper on your PI

Now make sure you have completed these steps denoted in klipper3d / Measuring_Resonances.html
1. sudo apt update
2. sudo apt install python3-numpy python3-matplotlib libatlas-base-dev
~/klippy-env/bin/pip install -v numpy

Now your Fysetc PIS should be ready to use and your PI/Klipper setup ready for the Resonance Compensation Input Shaping testing.

```

DaVinci-10, 2023-05-22:
BTW, I have learned you can now flash directly from kiauh. no need to press the boot button.
worked from kiauh wohoo
```Loaded UF2 image with 114 pages
Found rp2040 device on USB bus 1 address 13
Flashing...
Resetting interface
Locking
Exiting XIP mode
Erasing
Flashing
Rebooting device
[✓ OK] Flashing successfull!

###### Start klipper.service ...
[✓ OK] Start klipper.service successfull!```

Landsailor, 2023-06-15:
> BTW, I have learned you can now flash directly from kiauh. no need to press the boot button. worked from kiauh wohoo
> 
> ```
> Found rp2040 device on USB bus 1 address 13
> Flashing...
> Resetting interface
> Locking
> Exiting XIP mode
> Erasing
> Flashing
> Rebooting device
> [✓ OK] Flashing successfull!
> 
> ###### Start klipper.service ...
> [✓ OK] Start klipper.service successfull!```
> ```

**THANK YOU!**

This was the only way I could get the stupid PIS to work.

zgmorris13, 2023-08-02:
> just as a follow-up from this thread I was also having issues, then I found a comment from `Kris S.` in the [amazon product](https://www.amazon.com/dp/B0B8MMW2Z9/ref=cm_cr_dp_d_vote_lft?ie=UTF8&csrfT=hJhQyngdabosrjrDIT17tKW1hi05rm4BDse3C2g0etKIAAAAAGRhPWoAAAAB&reviewId=R2W18RGW1VIAX3#R2W18RGW1VIAX3) page that finally made it work here
> 
> ```
> 
> The Fysetc instructions on Github are not complete and unless you are a pro you may have issues getting it set up.
> 
> With the help from klipper discord these instructions helped me get it set up.
> 
> MAKE SURE YOUR KLIPPER IS UP-TO-DATE
> 
> Via SSH to your PI with Klipper installed
> 1. Change directory to Klipper
> 2. make menuconfig
> 3. set menu per image on Fysetc PIS Github
> 3.1 Do not enable extra low-level configuration options
> 3.2 Set Micro-controller Architecutre to Raspberry PI RP2040
> 3.3 Set communication interface to USB
> 4. Exit and save menu config.
> 5. sudo service klipper stop
> 6. make flash
> 7. sudo service klipper start
> Now the firmware 'klipper.uf2' should be created and located on your pi in the klipper/out directory
> 
> FTP into your PI
> 1. Navigate to the Klipper/Out directory
> 2. Download the 'klipper'uf2' file onto your pc desktop.
> 3. Exit and quit FTP
> 
> Now flash the firmware 'klipper.uf2' onto the Fysetc PIS
> 1. Press and hold the button on the Fysetc PIS
> 2. Connect the PIS with USB C to your PC
> 3. Release the button on the PIS
> 4. A folder named RPI-RP2 will pop up
> 5. Copy/Move the 'klipper.uf2' file from your desktop into the RPI-RP2 folder
> 6. Wait - the folder should close automatically
> 7. Unplug the PIS from your PC
> 
> Now set up PIS on your PI via USB
> 1. Press and Hold the button on the PIS
> 2. Connect PIS to your Raspberry PI
> 
> Now set up the firmware
> 1. Via SSH issue - lsusb
> 2. locate the device ID for the RP2040 -- i.e. Bus 001 Device 008: ID 2e8a:0003 Raspberry Pi RP2 Boot
> 3. Copy the ID i.e. '2e8a: 0003'
> 4. CD to your Klipper directory
> 5. issue command- make flash FLASH_DEVICE=2e8a:0003 << CHANGE TO YOUR DEVICE ID
> 6. You should see that the UF2 has been flashed
> 
> Now lets find the serial address for the Fysetc PIS RP2040 connected to your PI
> 1. issue command - ls /dev/serial/by-id/*
> 2. copy the address denoted with rp2040 -- ie. /dev/serial/by-id/usb-Klipper_rp2040_E66160F4235E4038-if00
> 
> Now lets create the PIS.cfg in klipper
> 1. Machine >> Create File >> name it PIS.cfg OR download it from Fysetc git up and upload the file
> 2. Copy contents from the Fysetc PIS file into your kipper PIS.cfg
> 3. Edit the "serial" line to have your path you copied from the prior steps for obtaining the serial address
> 4. Save/close/restart Klipper on your PI
> 5. Edit your printer.cfg to add a line that includes the PIS.cfg -- [include PIS.cfg]
> 6. Save/close/restart Klipper on your PI
> 
> Now make sure you have completed these steps denoted in klipper3d / Measuring_Resonances.html
> 1. sudo apt update
> 2. sudo apt install python3-numpy python3-matplotlib libatlas-base-dev
> ~/klippy-env/bin/pip install -v numpy
> 
> Now your Fysetc PIS should be ready to use and your PI/Klipper setup ready for the Resonance Compensation Input Shaping testing.
> ```

I don't understand the double flashing here? why are we flashing from the PC only to reset the device and flash again from the PI?

zgmorris13, 2023-08-02:
> **THANK YOU!**
>
> This was the only way I could get the stupid PIS to work.

I have followed every guide out there and still with no luck. I am running moon raker and fluidd from my raspberry pi. Would this method you're talking about work?

DaVinci-10, 2023-08-03:
You can DM me either here or on Klipper discord or discourse boards with same name. I can help walk you through it.

> I have followed every guide out there and still with no luck. I am running moon raker and fluidd from my raspberry pi. Would this method you're talking about work?

zgmorris13, 2023-08-04:
@DaVinci-10 Not sure how to DM here, but I sent you a request on Discord

jtenniswood, 2024-02-08:
Thank you @gabriellferreira, the 'make flash' was the missing step, copying the file doesn't seem to flash the device automatically as other guides have indicated.

_In case anyone struggles, you need to be in the cd ~/klipper/ for that bit to work._