FYSETC/FYSETC-PortableInputShaper #11: Copying klipper.uf2 to PIS - Permission Denied

houstnwehavuhoh, 2023-06-05:
Hello,

I have created and downloaded the Klipper.uf2 - after holding the program button, inserting USB, letting go, the PIS shows up as a device. The issue I'm having is when I attempt to copy the .uf2 file to it, it says "Permission Denied"

Any advice? TIA

DaVinci-10, 2023-06-13:
FYI, You no longer need to press the boot button, you can now flash the PIS directly from klipper (or KIAUH)

Superflyron, 2024-01-07:
I have exactly the same problem, I've already tried all sorts of things, you can only mount the firmware on the input shaper on it, but then you have to remount it again and again. But even then it doesn't work because clipper then gives an error message where it is described that either the cable or the 2040 chip is defective. The whole thing is really annoying, because I really spent a lot of time trying to get it to work.