FYSETC/FYSETC-PortableInputShaper #12: Unknown config object 'adxl345 usbadxl'  when using default PIS.cfg

Landsailor, 2023-06-15:
After fighting and fighting to get this thing to work, then finally using KIAUH to flash it, I still received this error when trying to start klipper:

`Unknown config object 'adxl345 usbadxl'`

Disabling that line, enabling the line without "usbadxl", then restarting the firmware fixed it.

DaVinci-10, 2023-06-17:
Guys, Please update your example config to complete the last change. if your going to call it by name in `[resonence_tester]` then it needs to be named that way `[adxl345 usbadxl]`
```
[adxl345 usbadxl]
cs_pin: PIS:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: PIS:gpio10
spi_software_mosi_pin: PIS:gpio11
spi_software_miso_pin: PIS:gpio12
axes_map: x,-z,y

[resonance_tester]
#accel_chip: adxl345
accel_chip: adxl345 usbadxl
probe_points:
    100,100,20
    ...
    ```

oligneisti, 2023-07-05:
I ended up with this and it seemed to work.

[mcu PIS]
# Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
serial: /dev/serial/by-id/XXXX

[adxl345]
cs_pin: PIS:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: PIS:gpio10
spi_software_mosi_pin: PIS:gpio11
spi_software_miso_pin: PIS:gpio12
axes_map: x,-z,y

[resonance_tester]
# accel_chip: adxl345 usbadxl
accel_chip: adxl345
probe_points:
    100,100,20 # an example

jakebullet70, 2023-08-31:
+1 on this - there is even a pull ready to be merged with the fixed code.

shiftingtech, 2023-11-29:
@GerogeFu  Come on man, it's embarrassing that something this simple has been broken in your examples for 6 months

Stepulin, 2023-12-04:
Hi,
can confirm that the suggestion posted by @oligneisti works!
However, for people who does not have a lot of experience with GitHub it can be a bit tricky beacause of the hastag symbols while coping the text.

`
[mcu PIS]

#Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
serial: /dev/serial/by-id/usb-Klipper_rp2040_E6611CB6970E5A24-if00

[adxl345]
cs_pin: PIS:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: PIS:gpio10
spi_software_mosi_pin: PIS:gpio11
spi_software_miso_pin: PIS:gpio12
axes_map: x,-z,y

[resonance_tester]

#accel_chip: adxl345 usbadxl
accel_chip: adxl345
probe_points:
  100,100,20 # an example
`

Pharthan, 2024-01-27:
I can confirm that the fix above (renaming "[adxl345 usbadxl]" or the value for "accel chip:" such that they match) works, and that as of 1/27/2024, they still haven't fixed the example config.