FYSETC/FYSETC-PortableInputShaper #13: ls: cannot access '/dev/serial/by-id/': no such file or directory

zgmorris13, 2023-08-02:
I have followed every guide there is to get this to work and still with no luck. I did all the troubleshooting from the amazon page. I am able to compile using the make command, copy the file to the PIS. I plug it in to the raspberry pi and run lsusb and see that its ID has changed to 1d50:614e OpenMoko, Inc. rp2040. However when I run

`ls -l /dev/serial/by-id/`

I get the error

'ls: cannot access '/dev/serial/by-id/': No such file or directory'

If I try to flash the PIS directly from the PI instead of my Mac I get the following.

`pi@fluiddpi:~/klipper $ make flash FLASH_DEVICE=2e8a:0003
  Building rp2040_flash
gcc -c -Wall -ggdb -I../rp2040/ `pkg-config libusb-1.0 --cflags` main.c -o main.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
In file included from main.c:12:
picoboot_connection.h:13:10: fatal error: libusb.h: No such file or directory
   13 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:17: main.o] Error 1
make: *** [src/rp2040/Makefile:69: lib/rp2040_flash/rp2040_flash] Error 2`

This is what I see when I run lsusb.
'pi@fluiddpi:~/klipper $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2e8a:0003 Raspberry Pi RP2 Boot
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub'

I've followed the troubleshooting guide and replugged and reset the pi so many times my brain is going to explode. I feel like the PIS should be renamed POS and I should be paid for the hours of my life wasted trying to get this thing to work. I even ordered a replacement to see if it was the device I had was messed up somehow, but I got the same exact results.

DaVinci-10, 2023-08-04:
feel free to DM me via klipper discord with same name. I can walk you through a few checks.

asteri-matia, 2023-11-28:
I just fixed it!!
I had the exact same issue and have just spent a stupid amount of hours reading about the inner workings of linux...
Paying it forward <3

[my setup]
Orange Pi Zero 3 - 4GB
meganukebmp build of MainsailOS (r1) || Version 1.2.1, based on Orange Pi OS Bookworm (1.0.2)
Kernel: Linux 6.1.31-sun50iw9
Klipper: Version: v0.11.0-303

STEP 00::
Firstly try running::
cd /
ls dev/serial/by-id*

If that doesnt work; watch this vid; 
   https://www.youtube.com/watch?v=UNGrYuZdwXk
and copy this file (in vid description but just in case the vid goes down in future)
   https://raw.githubusercontent.com/systemd/systemd/main/rules.d/60-serial.rules
to this location: 
   /usr/lib/udev/rules.d/

STEP 01::
Open a terminal::
ssh pi@pi.ip.address
Backup the old file first. I copied it out to my config file (accessible via browser gui- easy to manage files from any pc)
   cp /usr/lib/udev/rules.d/60-serial.rules ~/printer_data/config/60-serial.rules.old

Save the github file above to your pc as "60-serial.rules"
Use browser to open your klipper install - mine is [mainsailos.local]
Upload (60-serial.rules) file using machine tab on the left, into your [config] folder

Go back to terminal::
cd printer_data/config/
sudo cp 60-serial.rules /usr/lib/udev/rules.d/

REBOOT using web browser (top right on mainsail)

Open terminal
ssh pi@pi.ip.address

Try STEP 00 again.

If that fails try different usb[A-C] cable, many users have reported lots of cable types not working