FYSETC/FYSETC-PortableInputShaper #15: Invalid adxl345 id (got ff vs e5)

sudzwerg80, 2023-10-18:
Invalid adxl345 id (got ff vs e5) 
i got it working one time then no more changed usb cables with both configs this repo and nero 
what's the Problem reflashed it twice bought a second PIS .
Still dont get it working..

GaryB477, 2023-10-31:
Hi @sudzwerg80!
Dumb question, but are you perhaps using a mellow `Fly-ADXL345`? I just stumbled over the same issue with the exact same faulty id's. Since I am using the described model, using this config solved the problem for me:
```
[mcu adxl]
# Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
serial: /dev/serial/by-id/usb-Klipper_rp2040_<id>

[adxl345]
cs_pin: adxl:gpio9
spi_software_sclk_pin: adxl:gpio10
spi_software_mosi_pin: adxl:gpio11
spi_software_miso_pin: adxl:gpio12
axes_map: x,-z,y

[resonance_tester]
accel_chip: adxl345 
probe_points:
    150,150,20 # an example
```

Especially the `cs_pin: adxl:gpio9` seems different. Also, please be aware that you are remapping your axes twice 
```
axes_map: x,-z,y
axes_map: x,z,y
```

Maybe it helps :)

sudzwerg80, 2023-10-31:
thank you.
 i will test it! will report back

Romancc, 2023-11-23:
Changing from  `cs_pin: PIS:gpio13` to `cs_pin: PIS:gpio9` fixed my problem using the mellow fly-adxl345

sudzwerg80, 2023-11-25:
sorry for the late answer but it didn't work both pins cs_pin: PIS:gpio13  cs_pin: PIS:gpio9  doesn't work
 but 
`16:05:11 $ ACCELEROMETER_QUERY CHIP=adxl345
16:05:13 // accelerometer values (x, y, z): -444.123565, 9252.574275, 153.219100`
works

sudzwerg80, 2023-11-25:
sorry for the wrong information it works had to restart the Pi it works
thank you
this config works for me
[mcu PIS]
# Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
serial: /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0

[adxl345]
cs_pin: PIS:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: PIS:gpio10
spi_software_mosi_pin: PIS:gpio11
spi_software_miso_pin: PIS:gpio12
axes_map: x,y,z

[resonance_tester]
#accel_chip: usbadxl
accel_chip: adxl345
probe_points:
    110,110,20 # an example

sudzwerg80, 2023-11-25:
thank you both

closed 2023-11-25 by sudzwerg80