FYSETC/FYSETC-PortableInputShaper #16: two fysetc input shapers config

xenoton79, 2023-11-03:
If someone wants to connect two at the same time, here is the config.

`# Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
[mcu hotend]
serial: /dev/serial/by-id/***
[mcu bed]
serial: /dev/serial/by-id/***

[adxl345 usbadxl hotend]
cs_pin: hotend:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: hotend:gpio10
spi_software_mosi_pin: hotend:gpio11
spi_software_miso_pin: hotend:gpio12
axes_map: x,-z,y

[adxl345 usbadxl bed]
cs_pin: bed:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: bed:gpio10
spi_software_mosi_pin: bed:gpio11
spi_software_miso_pin: bed:gpio12
axes_map: x,-z,y

[resonance_tester]
#accel_chip: adxl345
accel_chip_x: adxl345 usbadxl hotend
accel_chip_y: adxl345 usbadxl bed
probe_points:
    100,100,20`

chilman408, 2023-11-10:
Hi @xenoton79,

Can you provide an example on how to do this same thing for 3 Fysetc Input Shapers?  I have a Sovol SV04, so I have X1, X2 and Y.

I'm at a lost because accel_chip_x and accel_chip_y are specific commands...  I'm wondering how to do this for a dual carriage.

xenoton79, 2023-11-10:
I'm not sure if I can help with this or if it even works that way.

According to my logic you only have one X axis. No matter how many extruders are attached to it.
So you only need two Fysetc input shapers.

However, I'm not sure what is referred to as X in your case and whether input shaping can even handle X1 and X2.

You have to test this yourself.
Otherwise, here is an interpretation of how it could work.

[mcu hotend1]
serial: /dev/serial/by-id/***
[mcu hotend2]
serial: /dev/serial/by-id/***
[mcu bed]
serial: /dev/serial/by-id/***

[adxl345 usbadxl hotend1]
cs_pin: hotend1:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: hotend1:gpio10
spi_software_mosi_pin: hotend1:gpio11
spi_software_miso_pin: hotend1:gpio12
axes_map: x,-z,y

[adxl345 usbadxl hotend2]
cs_pin: hotend2:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: hotend2:gpio10
spi_software_mosi_pin: hotend2:gpio11
spi_software_miso_pin: hotend2:gpio12
axes_map: x,-z,y

[adxl345 usbadxl bed]
cs_pin: bed:gpio13
#spi_bus: spi1a
spi_software_sclk_pin: bed:gpio10
spi_software_mosi_pin: bed:gpio11
spi_software_miso_pin: bed:gpio12
axes_map: x,-z,y

[resonance_tester]
#accel_chip: adxl345
accel_chip_x1: adxl345 usbadxl hotend1
accel_chip_x2: adxl345 usbadxl hotend2
accel_chip_y: adxl345 usbadxl bed
probe_points:
100,100,20`

chilman408, 2023-11-10:
Thank you for the quick reply.  

The SV04, the IDEX bed slinger, visually have 1 X-rail but have 2 independent belts and 2 independent motors on this rail and the 2 X heads move independently on X.

Unfortunately, I don't think accel_chip_x1 or aceel_chip_x2 are valid configuration parameters.