FYSETC/FYSETC-PortableInputShaper #17: Configuration for axes_map  with respect to mounting point

mysteryjeans, 2023-11-11:
Why axes_map is x,-z,y whats the significance "-z" before "y"?

Creo2005, 2024-04-21:
> Я бачу, що ніхто не відповів. Тож ось: друкована плата має трафарет, який показує кожну площину осі та який напрямок є «позитивним». Однак axes_map більше не потрібен для відповідності монтуванню в конфігурації та впливає лише на кольори у створених діаграми.. але щоб відповісти на ваше запитання, це порядок осі, а додавання "-" означає зворотні напрямки на певній осі.
> 
> Існує 24 різні комбінації карти осей залежно від того, як вона фізично змонтована (по 4 на кожній грані куба на аркуші). Раніше було важливо, щоб карта осі відповідала фізичному монтажу. Але тепер математика формувача вхідних даних враховує всі осі під час запуску інструменту формування вхідних даних, тому його більше не потрібно збігати.
> 
> Тож ви можете просто залишити його в спокої або змінити його на x,y,z за замовчуванням, або якщо ви хочете, щоб кольори на графіках збігалися, зробіть так, щоб він відповідав тому, що він фізично є у ваших налаштуваннях. Я створив шпаргалку .3mf файл, який допоміг мені визначити, яка axis_map відповідає орієнтації встановленої друкованої плати. Пишіть мені тут або на Discord, якщо хочете копію, оскільки я не можу опублікувати тут файл .3mf.
> 
> ![PIS_Axis_CheatSheet](https://private-user-images.githubusercontent.com/1100376/324253494-a13cf5a9-2bfe-43d8-b130-32042bc34eac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM3MDk5MDYsIm5iZiI6MTcxMzcwOTYwNiwicGF0aCI6Ii8xMTAwMzc2LzMyNDI1MzQ5NC1hMTNjZjVhOS0yYmZlLTQzZDgtYjEzMC0zMjA0MmJjMzRlYWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjFUMTQyNjQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWEwODg1ZTVmMjBiYmQ3ZWI4Njk3NmRjZDY3MmJjYjE5Y2EwZDRiN2FjNDg1NmVhZjBhMWEzNTNkYWFlYjFlYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.nvitQaovVmytU1hqN-G_cxDzY3rkMwVULyTVyQNlf6w)

can you make a zip archive?

DaVinci-10, 2024-04-23:
I'll try.. 
[FYSETC-PIS-mount-cheat-sheet-PS2.5.2.3mf.zip](https://github.com/FYSETC/FYSETC-PortableInputShaper/files/15084117/FYSETC-PIS-mount-cheat-sheet-PS2.5.2.3mf.zip)